NAK3DDesigns/White-Knight #1: Update BOM to use markdown files

ArmoredBlood, 2019-04-03:
the current BOM is awkward to read since it needs a spreadsheet application just to open. I suggest consolidating the BOM and the order pdf's into a single readme.md that contains all the info of all the original spreadsheet and pdf's.

servant74, 2019-04-03:
Try this, the link has a PDF you can check out.  I don't have update privleges.  This is a conversion using Google Docs.

[White Knight BOM.pdf](https://github.com/NAK3DDesigns/White-Knight/files/3036523/White.Knight.BOM.pdf)

ArmoredBlood, 2019-04-03:
If you check out #2 I added the connector as an image and it's embedded in the BOM readme.md file. This way you don't need to download anything just to view the contents, it can all be viewed directly in your browser

NAK3DDesigns, 2019-04-04:
This issue has been merged with master.  Closing issue... PR#2

closed 2019-04-04 by NAK3DDesigns