NAK3DDesigns/White-Knight #7: Claim STL files you want to convert to STEP / Fusion 360

3d-gussner, 2019-04-06:
Just an idee:
Wouldn't it make sense if people claim ONE STL file at the time they are converting to STEP / Fusion 360? Or making a conversion status list:

 | STL | Part of STL | Who | Status | ETD |
 | --------------------------- | -------------- | ---------- | ------ | ------- |
 | 1-All_Belt_Clams V2.stl | clam nr. 1 | @3d-gussner | Done  | 4 april 2019 |

This way we could prevent double conversions and hopefully get it done faster.

What do you think?

NAK3DDesigns, 2019-04-06:
I love the idea, but I know about as much about GitHub as I do about Fusion.. lol  I'm just a little out of my element..

dsk001, 2019-04-06:
I will start working on the Belt stepper mount Pt1 V4.stl

NAK3DDesigns, 2019-04-06:
> I will start working on the Belt stepper mount Pt1 V4.stl

Please check the fusion 360 folder in the development branch before selecting files till we get a better system in place.  I do believe that the Belt Stepper Mount Pt1 V4 has already  been done.

3d-gussner, 2019-04-07:
@NAK3DDesigns As github Owner you can edit this issue and update the claims

3d-gussner, 2019-04-07:
> I will start working on the Belt stepper mount Pt1 V4.stl

any ETD or Info i can add to your claim?

3d-gussner, 2019-04-07:
@NAK3DDesigns added a column Reviewed, so you can say it is done or need some rework.

EXOgreen, 2019-04-07:
I've got cooling ducts v4 left and therefore right as well

dsk001, 2019-04-07:
@3d-gussner since it is already done, I didn't do it. No use duplicating work already done.

NAK3DDesigns, 2019-04-07:
Thanks everyone, bare with me if it takes me a bit to get a rhythm going I’m still learning how to use Github but I don’t want to slow you guys down.

reFlex-lab, 2019-04-07:
I guys,
I'm totally New to github actually i came because of this project. But i am really good at fusion 360. I would like to help.
how Can I do that ?

EXOgreen, 2019-04-07:
@reFlex-lab the way I have found to work with this (also new here) is to fork this github repository. This creates an exact copy of the repository under your account. Once you have a file that you gave recreated in fusion, upload both the .F3D and .STEP files to the development branch of your repository (so the parts can be tested before being sent to the master). You can then make a push request to this projects development branch that includes your updated files. Don't have to time right now to write up a detailed description of where all the buttons are but hopefully this gets you started!

3d-gussner, 2019-04-07:
> I guys,
> I'm totally New to github actually i came because of this project. But i am really good at fusion 360. I would like to help.
> how Can I do that ?

Claim here a part with ETD if possible

- on the github website `https://github.com/NAK3DDesigns/White-Knight/tree/Development` hit the `Fork` button

On Windows you can install git-bash

- create a folder "White-Knight"
- open git-bash
- change to that folder
- enter `git clone https://github.com/<your github user>/White-Knight.git master`
- enter `cd master` you should now see `.../White-Knight/master (master)`
- enter `git remote add upstream https://github.com/NAK3DDesigns/White-Knight.git`
- enter `git remote -v` to see if it worked
- enter `git worktree add -B Development ../Development`
- enter `cd ../Development` you should see `.../White-Knight/Development (Development)`
- enter `git pull`

To update your git repository with the root one:
- open git-bash in the` .../White-Knight/Develompent` folder
- you should see `.../White-Knight/Development (Development)`
- enter `git fetch --all`
- if you see git new things in the upstream
   - enter `git merge upstream/Development`
   - enter `git push` to update your git repository

To work on new part:
- Claim the part in this issue
- open git-bash in the` .../White-Knight/Develompent` folder
- you should see `.../White-Knight/Development (Development)`
- enter `git fetch --all`
- enter `git merge upstream/Development` to be up to date with the root
- enter `git push` to update you web git repository
- enter `git worktree add -b <what you are working on> ../<what you are working on>`
- change to that folder and do your Fusion360 magic
- as soon you finished the design
- open git-bash in the `.../White-Knight/<what you are working on>`
- you should see `.../White-Knight/<what you are working on> (<what you are working on>)`
- enter `git status` this should show your changes
- enter `git commit -a` to commit your changes, write something usefull
- enter `git push` if that hasn't been done before you will get a `git push ?????` suggestion
- copy paste that one 
- now you should see on your web git hub as a new branch <what you are working on>

Create a Pull request
- on weg github change to <what you are working on> branch and hit `Compare`
- if you are happy what is shown press `Create new pull request`

Having individual worktrees prevents pull/push/merge issues

Hope that helps

reFlex-lab, 2019-04-07:
@3d-gussner 
Thank you for the detailed description.

3d-gussner, 2019-04-08:
Updated the git commands...
Claimed 10-Bowden_Cable_Clip_V1.stl
created Pull Request for Fusion360 archive and STEP files, would be great if somebody could review it before it is merged

ianespana, 2019-04-08:
Count me in! I can only make Solidworks files, however the STEP format should suffice, unless anyone has issues with that?
I'll take:
Mosquito & E3Dv6 Mount V7.stl
Mosquito & E3Dv6 clamp and cooling fan mount V4.stl

EXOgreen, 2019-04-09:
Fan ducts are done and pushed, parametrics in the F3D file are kinda shit though >.> I'll take Belt stepper mount Pt2 V3 now.

3d-gussner, 2019-04-09:
> Count me in! I can make Solidworks files (I know F360 but get around much better in SW), however the STEP format should suffice, unless anyone has issues with that?
> I'll take:
> Mosquito & E3Dv6 Mount V7.stl
> Mosquito & E3Dv6 clamp and cooling fan mount V4.stl
Welcome, Fusion 360 archive files are prevered as most users can use it for free and we can change sketches if needed.
@ianespana Claim added

3d-gussner, 2019-04-09:
> Fan ducts are done and pushed, parametrics in the F3D file are kinda shit though >.> I'll take Belt stepper mount Pt2 V3 now.

Claim added

reFlex-lab, 2019-04-09:
I will Start with 1-Display_Support_Endcaps.stl

paulbuzzby, 2019-04-09:
I'd like to help out with some of the conversions. I'm distinctly average at fusion but should be able to help. What is the process people are using to do the conversion?

NicoSchlueter, 2019-04-09:
1-All_Belt_Clams V2.stl clamp nr.3 &
1-All_Belt_Clams V2.stl clamp nr.4

claimed, done, and pull-requested.

3d-gussner, 2019-04-09:
> I'd like to help out with some of the conversions. I'm distinctly average at fusion but should be able to help. What is the process people are using to do the conversion?
@paulbuzzby Just download the Fusion360 archives and you can see the different approaches.
Here is how i normally start:
- Import STL
- Move it to Origin
- Disable History
- Convert Mesh to BRep
- Enable History
- Try to cleanup some faces to get smaller BRep
- Start NEW sketch an reverse engineer it

3d-gussner, 2019-04-09:
@reFlex-lab and @NicoSchlueter claims added

3d-gussner, 2019-04-09:
Sorry hit the wrong button :-)

NicoSchlueter, 2019-04-09:
1-CBA bottom Cover V3 Full.stl &
1-CBA top Cover V5 part 1of3.stl &
1-CBA top Cover V5 part 2of3.stl &
1-CBA top Cover V5 part 3of3.stl &
1-CBA top Cover V5.stl &
1-CBA_slide_tray_assembly_v11.stl &
-Display_Support_Endcaps.stl short one &
-Display_Support_Endcaps.stl long one 

Claimed, Done, and should be added automatically to my previous pull request.

3d-gussner, 2019-04-09:
@NicoSchlueter just saw it thanks
@reFlex-lab If your are close to finish just make a PR we can then see which design is "nicer"

NicoSchlueter, 2019-04-09:
I would like to claim:
2-4080 c-beam endcap V5.stl
2-4080 c-beam endcap wider v1.stl
2-Side_Support_Bottom_Brace.stl
3-Pulley_Bearing_Mount_V2.stl
switch enclosure V4.stl

3d-gussner, 2019-04-09:
I am amazed that in less a week we already got 29 claims and results of 44 todo's.

You guys are awesome, love the community!

3d-gussner, 2019-04-09:
> I would like to claim:
> 2-4080 c-beam endcap V5.stl
> 2-4080 c-beam endcap wider v1.stl
> 2-Side_Support_Bottom_Brace.stl
> 3-Pulley_Bearing_Mount_V2.stl
> switch enclosure V4.stl

Man you are on a run :-) 
Can you please create a different github worktree and PR, otherwise your PR maybe in review and then it is being updated with new designs. This way we can give others time to review a PR one by one

NicoSchlueter, 2019-04-09:
> Can you please create a different github worktree and PR, otherwise your PR maybe in review and then it is being updated with new designs. This way we can give others time to review a PR one by one

I will try to do that, will have to look up how to do that thou.

3d-gussner, 2019-04-09:
@NicoSchlueter https://github.com/NAK3DDesigns/White-Knight/issues/7#issuecomment-480613712

NicoSchlueter, 2019-04-09:
2-4080 c-beam endcap V5.stl
2-4080 c-beam endcap wider v1.stl
2-Side_Support_Bottom_Brace.stl
3-Pulley_Bearing_Mount_V2.stl
switch enclosure V4.stl

Done and I figured the tree-thing out. Didn't read that initially since it started with "On windows", Sorry.
Please delete the initial pull request, everything has now been separated into many requests.

NicoSchlueter, 2019-04-09:
I would like to claim
1-hotend_carriage_v6.stl

Could you tell me what thread that is?

ianespana, 2019-04-10:
Completed Mosquito & E3DV6 Mount V7,

But man, this "Mosquito & E3Dv6 clamp and cooling fan mount V4" piece is being a bi***! The angles are really weird, xD

3d-gussner, 2019-04-10:
Updated list added claims @NicoSchlueter & @ianespana .

12 of 44 left to covert.

I claim 1-Peeler.stl :-)

paulbuzzby, 2019-04-10:
I'd like to claim the Plug Mount

3d-gussner, 2019-04-10:
> I'd like to claim the Plug Mount

@paulbuzzby 
Added claim to large and short 1-plug mount.stl

3d-gussner, 2019-04-10:
I claim 1-slide tray end cover v4.stl

NicoSchlueter, 2019-04-10:
1-hotend_carriage_v6 almost done, just need to know what thread that is.

NicoSchlueter, 2019-04-10:
I claim:
1-Stepper_Mount_Left_V9.stl
1-Stepper_Mount_Right_V9.stl
1-hotend wire cover V3.stl
1-tensioner mount bracket.stl

3d-gussner, 2019-04-10:
> hotend_carriage_v6

https://github.com/NAK3DDesigns/White-Knight/tree/master/BOM can it be the 3/8 PTFE, and fittings???

3d-gussner, 2019-04-10:
Last chance to claim last 4 of 44:

1-Rear_Idler_Right_V5.stl
1-Rear_Idler_left_V4.stl
1-Side_Support_Top_Brace_Left_V2.stl
1-Side_Support_Top_Brace_Right_V2.stl

ianespana, 2019-04-10:
Give me those, 😄 

1-Rear_Idler_Right_V5.stl
1-Rear_Idler_left_V4.stl

3d-gussner, 2019-04-10:
So it is nearly done
- 24 conversions have been already pulled to development
- 11 conversions are in 7 Pull Requests
- 8 conversions are ongoing

I guess next step is to review all files and tine tune them. So weird angles are removed. Maybe optimized for 3d printing .... 

Let's see what @NAK3DDesigns want us to do next :-)

3d-gussner, 2019-04-10:
> Give me those, 😄
> 
> 1-Rear_Idler_Right_V5.stl
> 1-Rear_Idler_left_V4.stl

Added claim...

ianespana, 2019-04-10:
What I suspect we'll have to do is make the assembly in CAD, and then adjust all models so they fit together properly. I'm willing to do that

3d-gussner, 2019-04-10:
@ianespana I was planing to do the same :-) as we have nearly all Fusion360 files

NAK3DDesigns, 2019-04-10:
I guess just let me know when you think the files are ready for me to print and test fit as I won’t publish them to the Master till I have test printed them.

ianespana, 2019-04-10:
@NAK3DDesigns @3d-gussner ETA 1 hour, maybe a tad bit less. Working on the last piece

ianespana, 2019-04-10:
@NAK3DDesigns @3d-gussner Done, pull request added

NAK3DDesigns, 2019-04-10:
Wow!!!!!   Never did I expect all these files would be converted so quickly.  Thank you!!!! I’ll run the pull requests tonite.

ianespana, 2019-04-10:
Heh, that's the power of the community!
Now I only wish I had the money to assemble one, heh

NicoSchlueter, 2019-04-10:
I am now all done as well.

racenviper, 2019-04-10:
I have created the C-Beam's and the other extrusions templates in Fusion, where can I add the drawing to get the correct dimensions to finish these? Having these will allow to have a complete assemble printer in Fusion. I can add drawings to show the tapped holes on the end of the extrusions also. I planed to create the Cold and Heated plates also from the STL's, but the dimensions will only be a guess from the STL's.

paulbuzzby, 2019-04-10:
At the moment i cannot get git to behave itself. Will try again later but as I think they are all done mine is here https://a360.co/2P1VzE0 Hope it helps

paulbuzzby, 2019-04-10:
Might have made it work. You should have a pull request for my files.

EXOgreen, 2019-04-10:
I'll have my final part done tonight, it's a really easy part just alot of other stuff going on. I'm wondering if we have any pictures or diagrams showing where each type of screw goes? Then we can more accurately size the holes as well.

3d-gussner, 2019-04-11:
After double checking the STL files and converted / pull request, I found two missing files in the list:
- 1-Xaxis_Mount_Right_V10.stl
- 1-Xaxis_Mount_Left_V11.stl

I claimed them as it was me forgetting them to add to the list above.

@NAK3DDesigns sorry Not done yet :-)

3d-gussner, 2019-04-11:
@ianespana Do you think it would be hard for you to add Fusion360 versions of your file, with sketches  and history so changing them in the future might be bit easier? Or are you okay if somebody else does that?

NAK3DDesigns, 2019-04-11:
I added an Extrusion Template folder in the Fusion 360 directory for those converted files..

[41 more replies]

closed 2020-01-02 by NAK3DDesigns