NAK3DDesigns/White-Knight #12: Possible infringement of Stratasys patents

milkey-mouse, 2019-04-07:
I'm not a lawyer or anything, but the main reason I didn't experiment with infinite-volume/belt-based 3D printers before was Stratasys patents [US8282380](https://patents.google.com/patent/US8282380) & [US8287794](https://patents.google.com/patent/US8287794).

You have probably encountered these before, but is there a consensus, or hope, that these patents don't cover a printer with a slanted Z axis like this one?

This may be a potential barrier to commercialization (and I wouldn't put it past Stratasys's lawyers to go after individuals, either).

wjsteele, 2019-04-07:
I designed this mechanism specifically to get around their patents. In both patents, claim one fails this kinematic model because we are not positioning the extruder in the XYZ space... All the rest of the claims are dependent claims, so they do not apply.

Bill

NAK3DDesigns, 2019-04-07:
Thanks Bill..  I'm assuming my design is within the same constraints as yours..

wjsteele, 2019-04-07:
Yes, it definitely is.

Bill

milkey-mouse, 2019-04-07:
Nice!

closed 2019-04-07 by milkey-mouse