NAK3DDesigns/White-Knight #46: Fusion 360 file organization

EXOgreen, 2019-04-16:
How should the files in the Fusion 360 folder be organized? For example, I found the STEP file for the 40mm blower fans on grabcad, But I don't know where to place them. I added the Bulk hardware from McMaster car to its own folder, should we have a folder for electronics, bulk parts, and 3D printed parts so that we can find what we are looking for easier? what would that structure be? LMK your thoughts.

3d-gussner, 2019-04-16:
@EXOgreen check the Development branch, we added few folders

EXOgreen, 2019-04-16:
ooooh. I was just looking in the fusion folder. Ill move bulk hardware around as well. Thanks

closed 2019-04-16 by EXOgreen

3d-gussner, 2019-04-16:
I think as soon we Karl is happy with the Fusion 360 files he will move this folder somewhere under Printed parts or so.