NAK3DDesigns/White-Knight #49: C-beam cutouts STL vs Video dimensions

3d-gussner, 2019-04-17:
Hi Karl,

I am busy to update the C-beam Fusion 360 files using your dimension from the White Knight Assembly Part 2 video.

- 5.5mm holes
  - 198mm from front
  - 342mm from front
  - 763mm from front
  - holes are 10mm from the center or 30mm + 50mm from bottom
They are bit different from the STL but don't see an issue here.

- C-beam_1 electronics cutout
  - 210mm from front
  - 335mm from front
  - and 23mm from center or 17mm + 63mm from bottom

- C-beam_2 electronics (switch enclosure) cutout
  - 240mm from front
  - 300mm from front
  - and 18mm from center or 22mm + 58mm from bottom

I have few issues/questions:
The holes on C-beam_1 are 12mm and 7mm away from the cutout, should these have same distance? 

Maybe i am wrong but that looks wired to me as the "CBA_slider_tray (123mm x 35mm) " and "slide tray end cover" are sliding in the 570mm 2040 extrusions while having the 125mm x 36mm cutout not in the middle of the 4 holes.
Should it be centered as it is on the C-beam_2?
Changing the cutout to 207.5mm and 332.5mm awould center it between the holes.

On the other hand the C-beam_2 it looks good, as the Switch cutout (60mm x 36 mm) is 42mm from the holes on both sides /centered.

NAK3DDesigns, 2019-04-17:
I see what you are saying...  the weird thing is I'm checking the measurements of the prototype and they still measure out to what I listed and it appears centered..  I'm confused at this point..  I would say go with 207 and 333mm.

NAK3DDesigns, 2019-04-17:
just watched my video and it appears I did need to file down once side more than the other so I'd go with 7mm on each side..

NAK3DDesigns, 2019-04-19:
@3d-gussner you'll be happy to see I've added a Poll note to the part 2 video at 4:21 correcting the opening measurements to 207mm and 333mm..

3d-gussner, 2019-05-21:
done and pulled

closed 2019-05-21 by 3d-gussner