NAK3DDesigns/White-Knight #50: I could really use your help!

ludozen, 2019-04-19:
Hi there,
I really like your design of this printer! My name is Ludo, I am currently doing a graduation project at the TU Delft, the Netherlands, in which I am trying to design a 3D printer for production of medical devices in Africa. One of the main features I want to integrate in this design is a conveyor belt system, very similar to your design, to enable continuous printing. I am about to start building a first prototype, and had some questions that I was hoping you could help me with.

Do you experience any warping issues? I can imagine that cooling of the bottom layers will cause shrinkage, and could curl op the belt. If not, how did you solve this issue? What type of belt are you using? And where did you purchase this? I am having a hard time finding the right materials. Currently I am looking into a belt made from stainless steel (0.5mm thickness).

You can also contact me by email: ludolambers@gmail.com. Or we could have a quick skype call.

I would be very grateful for any help you can offer!

Here's a picture of a very quick sketch of my current idea. I'll be upgrading a Prusa MK3.
[image]

Best regards,
Ludo

wjsteele, 2019-04-19:
Be careful... that specific style of printer is covered under Stratasys' patents.    Here is a quote from their patent, "In yet another example, the conveyer 104 may, by movement of the sheet 118 of material, control movement in one axis (e.g., the y-axis), while the extruder 106 moves in the z-axis as well as one axis in the plane of the sheet."  This sentence is describing the exact configuration of your proposed printer.

My way of getting around this was to alter the definition and eliminate the Cartesian requirement they imposed... i.e., XYZ positioning.  We only position in the head in the X and Y axis... Z is a "virtual" axis represented by the result of a calculation involving the requested Z as well as the physical Y axis... a value I call Z' (Z Prime = f(ZY))

Bill

You can also contact me by email: ludolambers@gmail.com<mailto:ludolambers@gmail.com>. Or we could have a quick skype call.

[first concept sketch]<https://user-images.githubusercontent.com/24191430/56425941-9121eb80-62b6-11e9-8084-5d29cebdaaa4.png>

NAK3DDesigns, 2019-04-19:
I had some minor warping issues when I was using a thinner sheet belt, but since going to a 6mil or .15mm thick belt I haven't seem any warping yet..   I think you will find that a .5mm thick belt will be too thick.  I am then covering my belt with a custom sheet of Buildtak.  Hope this helps..  Links to both companies are below..

https://www.belttechnologies.com/
https://www.buildtak.com/

ludozen, 2019-04-19:
Thanks for your quick replies! It's a shame finding out that Stratasys' has patented this solution, thanks for pointing this out Bill. 

Ah I see I made a typo, I am intending to use a .1mm stainless steel belt, not .5mm. Good to know you do not experience warping with this.

madhunm, 2019-04-22:
@ludozen check out: https://hackaday.io/project/114738-automatic-infinite-3d-printer

ArmoredBlood, 2019-04-22:
another thing to check out @ludozen is forum.beltprinters.com. I'm trying to build a community of people that want to discuss and learn more about infinite style printers.

EXOgreen, 2019-04-22:
The owner of that hackaday project is unreachable and the instructions are super unclear. This project was 1000 times more organized from day 1.  I'll have to join the belt printers forum! sounds like my kinda place. :D

NAK3DDesigns, 2019-04-22:
@EXOgreen thanks..  Though this project is better because the community and people like you are helping to make it so...

ludozen, 2019-04-23:
Yeah my interest into this type of printer was actually sparked by the hackaday project. Just like @EXOgreen, I reached out to him, with no response. Thanks for your advice guys! I digged a little deeper, and actually came to the conclusion that this type of printer might not be the best fit for the context of hospitals in Africa. 

If you're interested, this is my new plan:
So the goal is to develop a low cost, low maintenance printer, that is able to have a high volume output (in terms of printed parts), with the least amount of human interference. It seems like this belt type printer will greatly increase costs, and overall challenge of the project. I have now decided to try and develop a part ejection system with less complexity, using some form of a wedge, or tool to knock prints of the bed. I'll be using Anycubic's Ultrabase bed, combined with a custom built active cooling system. My theory is that if I can cool the bed down fast enough, the parts can be pushed off with little force.

EXOgreen, 2019-04-23:
This is an idea I have seen and heard about in a few places before. It is a good starting point. A little isopropyl alcohol and compressed air should cool the bed pretty fast and not impact the print surface. Then it's a matter of creating a wall/wedge for cheep while also figuring out how to keep it from breaking parts if the parts are stuck to the build plate too much. Finally you'll need a queue system for ejection and the next job. I recommend octoprint with the print queue plugin. It allows you to specify custom end-of-print gcode for bed ejection and then starts the next job.

NAK3DDesigns, 2020-01-02:
Closing due to over 6 months of inactivity..

closed 2020-01-02 by NAK3DDesigns