NAK3DDesigns/White-Knight #51: Drip tray rework..

NAK3DDesigns, 2019-04-23:
Hi Guys, I had to make a change to the Build Plates see (v3) basically I offset the center 400mm area to the right.  By doing this the drip tray will now be too small to fill the void between the main 4080 extrusion and the buildpate..    I'm currently busy getting a mockup kit pieced together for PS to use for sourcing kit components and creating assembly instructions.  If anyone want to take a crack at stretching it out 5mm more, I'll test fitment one I receive the newly cut plates..  If you look at it from the top it should just be a matter of carrying the diagonal edges out 5mm further..  If there are any question let me know..

EXOgreen, 2019-05-15:
Can you post a picture of the area you are talking about? I'm not too familiar with where all the parts fit when its all assembled.

NAK3DDesigns, 2019-05-15:
[image]

here is a pic of the drip tray.. the area marked out in red would be the new size..  basically going 5mm further into the plate.  I have shifted the buildplate 5mm to one side to allow for a larger off belt area.

EXOgreen, 2019-05-16:
ok, that makes sense now. I'll get that updated. And maybe see if its angle is the same in both cad designs.

NAK3DDesigns, 2019-05-16:
awesome.. thanks

EXOgreen, 2019-05-16:
see pull request #52 and let me know if it needs work

EXOgreen, 2019-05-20:
@NAK3DDesigns just confirming you saw this

NAK3DDesigns, 2019-05-20:
Yes, I did.   Sorry just had a crazy weekend.

NAK3DDesigns, 2019-05-25:
just merged the request..  hopefully here soon I'll get time to start test printing all these fusion files...  Thanks again..

NAK3DDesigns, 2020-01-02:
Done!!!  Closing ticket

closed 2020-01-02 by NAK3DDesigns