NAK3DDesigns/White-Knight #53: Kits?

paulbuzzby, 2019-06-25:
Hi, This is probably the wrong place to ask but I saw somewhere about kits being made for this. Not only am I keen to acquire a kit with all the non-3D printed I would be extremely interested in helping out either sorting or helping to distribute these. I live in the UK and I know that shipping from the US is just eye-watering expensive to elsewhere in the world. 

Paul

NAK3DDesigns, 2019-06-25:
Hi, thanks for your interest in my design.   I would recommend contacting Printed Solid directly as they are going to be producing the kits.

paulbuzzby, 2019-06-25:
Thanks,

A different question along the sames lines. Has anyone put together places outside of the US that the Non-3D printer parts can be ordered?

NAK3DDesigns, 2019-06-25:
Yes, I believe ScaldaYT has pieces together a UK source list.

NAK3DDesigns, 2020-01-02:
*******UPDATE*******
I was just informed that due to legal reasons Printed Solid will NOT be producing kits nor will they even be able to carry the belt materials..  They do plan on building a White Knight group on the main page that will contain all the parts that they carry used in making a White Knight..

Sorry for the bad news..

wjsteele, 2020-01-03:
Legal reasons? Do they think there is IP stopping it?

NAK3DDesigns, 2020-01-03:
I don’t know all the details but I was told their patent lawyers advised against it.

peacho17, 2020-01-03:
I am currently working on a version that uses industrial strength aluminium foil for the bed rather than the belt. Could this bypass the patent infringement concearns?

It works out way cheaper too. The foil is near 1mm thick and is $10 for a roll 45cm wise and 30m long.

Regards
Mark Pietsch

NAK3DDesigns, 2020-01-03:
I'm not certain where the patent infringement concerns are but I'm guessing any printer that prints on a belt would be the issue..  
    Also with the aluminium you may have issues with it stretching and curling..  that is the reason why I use fully hardened stainless steel..  I tried cheaper stuff before and had that issue.

closed 2021-03-23 by NAK3DDesigns