NAK3DDesigns/White-Knight #54: Request:  Update the main Readme file with a list of printer capability specs

quarky42, 2019-08-15:
When you have time, would you please update the main Readme file to include a rundown of the various specs to include (but not limited to):

Layer Height Range (min to max)
Build Volume
Min and Max Travel Speeds
Filament Size
Max Bed Temperature
Max Hot End Temperature
Overall Weight
Overall Dimensions
Anything else you can think of that would be neat to see at a glance?

I've gotten the answers to many of these things from various places, but it would be nice to have them all in one spot (and to verify what I've heard in videos and pulled from other places is correct.)

Johannp02180, 2020-03-25:
I think it would help if you answered your own question and posted the specs you found and then he could easily verify them. Also, a lot of the specs depends on what hotends you buy and bed. You could easily lookup the hotend he used and find the max temp.

quarky42, 2020-03-25:
I had a suggestion for how the information available could be improved.  He could take it or leave it.  I wasn't trying to waste his time any more than necessary having a discussion about it.  Me guessing at the parameters when he should be familiar with them isn't helpful.

Perhaps if you had lead by example I'd see better what you're talking about.  A couple of obvious parameters don't address the specifications as a whole being lacking.

Johannp02180, 2020-03-25:
I see, if you watch the video by 3Dprintingnerd, you'll see even he himself isn't 100% on the specs, cause he hasn't tested the printer to its limits yet. I apologize if my previous comment came down hard.

quarky42, 2020-03-25:
Thank you for reconsidering your tone.  I did take it as pretty aggressive, but if it wasn't meant as such then I can accept that I was mistaken.

Yes, I did notice that he wasn't sure in that video but then in a later video he tests it and gets pretty close to the limit. That was 7 months ago.  I had hoped that at some point he'd consider adding the information.  Even if it wasn't complete at first, it could always be added over time.  

I figure the creator was the best place to start rather than trying to piece it together and guess.  I was particularly interested in the whole step positional accuracy vs the microstep accuracy.  Those two parameters would have been interesting to see.  Even if I had to calculate the effect of the triangular ZY axis on it, it would be neat to see how it compares to some other systems.

Johannp02180, 2020-03-25:
Yeah I agree it would be great to have that info and I didn’t mean to be aggressive. I was just thinking that having your info would be great for other users out there.

NAK3DDesigns, 2020-03-25:
Lets all play nice... ;) Sorry I haven't made any updates on the info requested but to be honest I'm not sure of the best ways to benchmark the printer to give accurate numbers.. Can it move at over 100mm/sec sure but at what loss of print quality, I do most of my prints at 30mm/sec with moves at 120mm/sec.  I've used nozzles from ,4 to 1.0 succesfully  but am still dialing in settings...  Sadly this is just a side hobby for me and I have many other projects I'm trying to work on as well..  To be honest I was kinda hoping I would publish what I had and the community would start building and improving my design because like I said it isn't perfect..  For example I've already given up on using the bowden design of the White Knight and switch mine over to a direct drive setup like I put on the Squire because I wasn't able to get a working slicer profile that eliminated the pressure feedback issues associated with such a long bowden run..  

Sorry this turned into a bit of a rant..  I'm just not a highly technical guy..  I'm good mechanically and decent with firmware and settings but I'm no Chris Riley with all the in depth details..  I may have gotten lucky that it works at all.. lol

quarky42, 2020-03-25:
I don't think you give yourself enough credit.  Maybe focusing on the kinematics and documenting them isn't the fun part, but it's clear you have a good eye for detail otherwise you wouldn't have had the success you've had.   There is no luck in complex system design.  There are just variables and managing those variables, and you've done a great job.

You highlighted a really important issue about movement speed.  A fair spec would be to say both normal print speed range and normal fast travel speed.  If the system allows for 120mm/s travel then print speeds over 30mm/s are possible with the right cooling and slicer settings.   As a baseline 30 and 120 are good data points.

What does 1 full step of the stepper motor equate to in movement in X?  How about it. ZY?  This info could inform layer height choices.

I'm not familiar with the squire.  I haven't caught up on the latest videos.  If I were building a similar setup to the White Knight, I'd be inclined to use the remotely driven / direct drive Zesty Nimble extruder.  That gets the weight of the motor off the moving carriage so less momentum but still direct driven through a torque cable.  What did you do on the squire?

NAK3DDesigns, 2020-03-25:
Thanks,
I think a lot of the info you are looking for is contained with in the Config.g file of the duet configuration.. 
 https://github.com/NAK3DDesigns/White-Knight/blob/master/Duet%20Config%20Files/sys/config.g

Current micro-stepping is set at 16x and config is as follows..  X=160, Y=160, & Z=208.75 so in full steps that would be X=10, Y=10, Z=13.047..

On the Squire I moved the Bondtech BMG and a pancake stepper to the top of the extruder carriage with approx 150mm of bowden between them..  I'm told it is similar to how the Prusa is set up.

I haven't really done any videos on the Squire yet (kind got burned out doing all the videos for the White Knight)  Joel did do an interview at ERRF where I covered most of the changes.  Here is the link to my portion of the interview.. https://youtu.be/bekwCnl9BwY?t=1343

Hope this helps a bit..