NAK3DDesigns/White-Knight #55: Belt Group Buy

JoshuaMorris, 2020-04-27:
Hi,

[Tj says he's done leading the group buys](https://infinityprinters.org/t/group-buy-endless-belt/67). I missed the last one and am willing to lead a group purchase to get a pair myself. If you're interested in joining in, comment on this issue and we can group up. I will keep this issue updated as we get enough people.

wjsteele, 2020-04-27:
I'm up for two!

madhunm, 2020-04-27:
@JoshuaMorris I would like one set please

sorren4, 2020-04-28:
How low does it go based on volume?  Is the pricing tiered?

AzonicTech, 2020-04-29:
I would like to get 2 of them. Is anyone putting together an excel sheet that we can add our name to? The one on the link will not let me add my name.

reFlex-lab, 2020-04-29:
I think it's already too late

JoshuaMorris, 2020-04-29:
@AzonicTech @reFlex-lab I'm working on setting up a group buy, I will be putting together a form/spreadsheet once I collect from everyone who wants to join me in doing a purchase. I'm just waiting a little to allow for other who might be interested in joining in.

spyle, 2020-05-15:
I'm interested too. I'm just now looking into building one and I really think I want the Squire. Are the belts used the same size on the squire and white knight?

NAK3DDesigns, 2020-05-15:
> I'm interested too. I'm just now looking into building one and I really think I want the Squire. Are the belts used the same size on the squire and white knight?

No, the Squire belt is almost half the size of the white Knight belt.

spyle, 2020-05-15:
@NAK3DDesigns Wheres the best place that I can bug you or a group with some questions?

NAK3DDesigns, 2020-05-15:
> @NAK3DDesigns Wheres the best place that I can bug you or a group with some questions?

NAK3DDesigns@gmail.com   First question is free additional are $5 per.. lol just kidding

cr10guy, 2020-05-19:
How do I sign up to buy I’ll take 4

JoshuaMorris, 2020-08-06:
After some consideration, this doesn't seem feasible for me to facilitate a group buy. I cannot at this time handle the redistribution of belts to everyone that wants to be a part of it. I would only be able to help those folks who could come and pickup their belts in person (Sacramento, CA).

tbedore, 2020-12-08:
@cr10guy not sure if you bought a belt or not but the pricing doesn't go down much from 4 belts i'd suggest for you to just buy the 4 yourself. And to everybody else I was the one who coordinated the first 2 group buys. They aren't really feasible unless you are driving to pick it up. For the belts, you pay for the belt in addition to the shipping cost from belt technologies to the person who is re-shipping them. Those boxes are not that small, 20x20x20 boxes, so you're paying for the repacking costs. And then on top of that the person who is coordinating the group buys, I did them through invoicely and square payment services. Those take a small percentage for using their services. So when looking at the price given to you by me, it is belt+shipping (belt to me) + shipping (belt to you) + shipping materials + CC processing fees. I looked at the numbers and after all of that, depending on your location will determine how close it is to price from belt technologies. I get away with some dirt cheap shipping, but they are located in MA. I would be willing to do another one if anybody is interested but I'm just putting this out there and being upfront about it.

tbedore, 2020-12-08:
Important to note also, buying 2 different size belts will not make the price go lower. The price per belt goes down because of their initial setup costs ($220 per size), so when the price per belt goes down that is solely because that initial setup cost is being split across all the belts. But say you got a squire belt and a white knight belt, you are paying for the setup cost on both of them, it doesn't get tiered together. You can reply to me here, or shoot me an email at ntwyoutube@yahoo.com i'd be more than happy to discuss purchasing belts. 

-TJ

tbedore, 2020-12-09:
If anybody is interested in another group buy of the belts, please keep in mind what I have said above, take a look at the [blog post I made on the infinity printers forum](https://infinityprinters.org/t/group-buy-endless-belt/67/51?u=tj_bedorel) and if you are interested after all of that feel free to add your details to the [google sheets](https://docs.google.com/spreadsheets/d/1kebBYrp085TU6l2bKg5meDZd1eZV_KM43DCM--KY5J4/edit?usp=sharing) that I have created . Happy Building

-TJ

Helvensteijn29, 2021-03-01:
Hi tbedore, kan i still sing up for a belt?

NAK3DDesigns, 2021-03-23:
Closing issue..  Please open a new ticket for each group buy and close the ticket once the group buy closes.

closed 2021-03-23 by NAK3DDesigns