NAK3DDesigns/White-Knight #59: Source: Thermal Isolating Washers?

Hvest3, 2020-09-24:
Hey Karl!

I'm curious as to where you sourced your thermal isolation washers for the hot and cold plates.  I apologize if you've covered this elsewhere, but I couldn't find them in the BOM.

Thanks man!

NAK3DDesigns, 2020-09-24:
The smaller ones that go above the heatplate are 

M5 high temp washers | 10 | McMaster-Carr | 93785A214

As for the larger 15mm OD washers I used some G10 like this
https://www.ebay.com/itm/G10-Sheet-Knife-Handle-1-8-x-5-x-11-7-8-RHINO-GREY/114295707394?hash=item1a9c8e1702:g:G6cAAOSwI9xfBzw4
and then used a hole saw with a 5mm center bit..

Hvest3, 2020-09-24:
Ahhh, G10, that makes sense.  I already had the PEEK M5 washers, but that clears up the mystery. Thanks man!

closed 2020-09-24 by Hvest3