NAK3DDesigns/White-Knight #60: Source: Hot/Cold Plate Machining

Hvest3, 2020-09-24:
Hey Karl,

Curious as to who you used to machine your hot/cold plates.  Been having trouble getting someone to do it without technical drawings and that's well beyond my CAD expertise.

Any recommendations you can offer are greatly appreciated.  Thanks man!

NAK3DDesigns, 2020-09-24:
To be honest unless you really want the very clean cut look you could just have the plates precut to the outer dimensions and do the cutouts with a jigsaw and drill the holes with a hand drill..  Otherwise the STEP or F3D files should be good enough for mist shops..  Sadly I had my plates waterjetted by a friend that doesn't work at the same shop anymore..

Hvest3, 2020-09-24:
Helps to have friends in a CNC shop for sure!  I'll keep searching for someone who can do it from the STEP files.  Appreciate the help and the quick reply!

closed 2020-09-24 by Hvest3