NAK3DDesigns/White-Knight #63: some STL-files can't be downloaded

LDekor, 2020-11-17:
Hi. i have trubble downloading some stl files. something that can be fixed?

these are the files that cant be downloaded:

2-Roller_Insert.stl

2-Side_Support_Bottom_Brace.stl

3-Pulley_Bearing_Mount_V2.stl

1-Peeler.stl

NAK3DDesigns, 2020-11-18:
I've re-loaded all of the files and they appear to be working now..  If you have any future problems try downloading using the desktop app..  It seems to work more reliably..

LDekor, 2020-11-19:
thank you. i am planing on building your printer. i have one question. how many parts do i have to make of the diferent printed parts? is it just one of each?

NAK3DDesigns, 2020-11-19:
the number at the beginning of the file name is how many you need to print..  i.e. **2**-Roller_Insert.stl   so you print 2..

closed 2021-03-23 by NAK3DDesigns