NAK3DDesigns/White-Knight #65: Interests in Partial Kits?

tbedore, 2020-12-31:
Hello,

I was wondering if anybody would be interested in possibly a frame kit from LDO Motors that is already cut down to spec for the white knight and with all the holes drilled in all the extrusions, holes cut, ends tapped, etc. You will probably still need to drill and tap the holes for the linear rails as that is different for everybody unless I could manage to get LDO to do that but I doubt they'd be willing to.

There is a very high likelihood that I would be able to get a motor kit from LDO also with everything that we would need. They wouldn't need to have connectors on them we could do just plain wires on the end all prepped for soldering later on with the wiring.

I could also probably get the linear rails cut down to size from them with the blocks and everything and have that all in one kit.  I want to see if anybody would be interested in any of these kits. Again the same idea with the fees and everything.

LDO -> Me
Me -> You
CC processing fee from Paypal

I would like to know if anybody would be interested in this.  If you are please add your name to the [spreadsheet](https://docs.google.com/spreadsheets/d/1zxAisEYr6v_wFz1Z-v1xb3HcQ-9nKwCt-PpczsmUoD8/edit?usp=sharing) and select from the dropdown what you would be looking for. I'm going to send an email to Jason over at LDO and see if he would be willing to do anything like this.

StumbleNOLA, 2021-01-28:
Honestly I would be much more interested in being able to buy a working machine for a premium over the cost.

servant74, 2021-01-29:
> <> ... Jack

Agreed

VinnyCordeiro, 2021-01-29:
A working printer would be nice, but I'd buy even a full kit in a box if that save me the time to chase all the components.

closed 2021-03-01 by tbedore