NAK3DDesigns/White-Knight #66: Multinozzel for Beltprinters

Prototyp-L98, 2021-03-08:
Hello everyone,

im building on my own version of a Beltprinter with 4 Nozzels.

But the problem is that neither CrealityBelt or CuraBlackbelt have multinozzel
for Beltprinter support. Dont get me wrong its work perfectly fine with "normal"
Printers but not for Beltprinters.

If u cant imagine what i mean i added my Fusion Hotend build.
[Belthotend.zip](https://github.com/NAK3DDesigns/White-Knight/files/6103319/Belthotend.zip)

NAK3DDesigns, 2021-03-09:
I suggest contacting Stewart at Kiri:Moto   - sa@grid.space    He has been working on a beta belt slicer and has been very eager to add functions as they are requested..  He thinks the current configuration will work.  Here  is the link to it..  https://grid.space/kiri/

Prototyp-L98, 2021-03-09:
Thank you very much that helped a lot.
I never heard of this Slicer but it looks very promising.
I'll get right on testing.
Thank you

closed 2021-03-09 by Prototyp-L98