NAK3DDesigns/White-Knight #68: Blackbelt cura

LDekor, 2021-04-23:
Hi. i downloaded the firmware some months ago but have not tried the firmware befoure now. the program crashed when choosing nozle so i deleted the program. now i cant find blackbetl 3.6.0 and the update because they only hav the new 3.6.2 program. what shoud i do to make this work?

- Thomas

NAK3DDesigns, 2021-04-23:
Hi, To be honest Blackbelt Cura is very out of date, at this point I recommend downloading IdeaMaker..
https://www.raise3d.com/download/
Then I'd recommend downloading one of the CR-30 profiles and just adjust the machine dimensions as I used my White Knight setting to build the original CR-30 profile.
https://www.ideamaker.io/

GpMissi, 2021-04-25:
> Hi, To be honest Blackbelt Cura is very out of date, at this point I recommend downloading IdeaMaker..
> https://www.raise3d.com/download/
> Then I'd recommend downloading one of the CR-30 profiles and [...]

Hi, I'm doing some simulations to evaluate the project. Slicing with Ideamaker I see that it is unable to recognize bridges, is there a solution for this problem? Thank you

(sorry my english, i use google translate)

GpMissi, 2021-04-25:
[image]

NAK3DDesigns, 2021-04-25:
Sorry maybe it's the translation but I don't see the problem..

Karl Brown
NAK 3D Designs
"The possibilities are Infinite"
Youtube: youtube.com/c/NAK3DDesigns <https://t.co/Wi7S9iGhtI?amp=1>
Twitter: https://twitter.com/NAK3DDesigns

GpMissi, 2021-04-25:
I try to explain my concern. As you can see from the image, the thread of material will be extruded as a normal filling and not as a bridge, this means that it will not go to an increased speed and the ventilation provided for the bridges will not be activated.

GpMissi, 2021-04-25:
[image]

riempimento solido = solid fill 

Ponte = bridge

NAK3DDesigns, 2021-04-25:
oh ok, sorry I don't know anyway to fix that, you can try contacting
ideamaker and open a trouble ticket..  admittedly I haven't had much
problem with bridging though..

Karl Brown
NAK 3D Designs
"The possibilities are Infinite"
Youtube: youtube.com/c/NAK3DDesigns <https://t.co/Wi7S9iGhtI?amp=1>
Twitter: https://twitter.com/NAK3DDesigns

GpMissi, 2021-04-25:
OK thanks. I will try to contact Ideamaker technicians :-)