NAK3DDesigns/White-Knight #72: Parts list

Gigas-E, 2021-06-29:
Hi I'm very interested in building my own white knight and I am needing a parts list so I can start buying the plethora of parts required. 

Thank the Lord I was blessed to get in on the belt group buy and already have the hardest part to get. All I require now is the rest. I trued to find a parts list with no success. If it's not too much trouble for you could you also include where you were able to source each piece I would be so grateful!!

Thanks for creating this awesome 3D printer!!! God Bless!

NAK3DDesigns, 2021-06-29:
You are already in the correct place..   everything you need to know is here in the github..  https://github.com/NAK3DDesigns/White-Knight/tree/master/BOM

midippel, 2021-06-29:
Hey @Gigas-E, I was going to be building the squire, but just had my first kid and don't really have time for this right now. I have everything already together if you'd be interested in buying it as a kit(minus the squire specific stuff). Email me at midippel@gmail.com