NAK3DDesigns/White-Knight #73: belts plastic

SeyedAli1998, 2021-07-26:
Hi Karl
What's the plastic material you used over the steel belt?
Is it a Neoprene Rubber Sheet?
Can you help me about it?

wjsteele, 2021-07-26:
His is covered with a custom BuildTak surface.

SeyedAli1998, 2021-07-26:
> His is covered with a custom BuildTak surface.

Don't you know the material name?
Unfortunately I couldn't buy from BuildTak

wjsteele, 2021-07-26:
You'd have to ask BuildTak for their material.

paulmn-web, 2021-08-01:
I custom built my belt on the printer bed - in essence the bed is an ideal set up for making laminate belts. I have a compression roller temporarily on the rigid portion of the bed, the first laminate layer is tape tacked together and then the layers are laminated onto that. Offset the laminate ends and I used Selley’s horizontal contact adhesive which is temperature stable to 130’c. 
Mine’s a 3 layer laminate with fibreglass or stainless mesh core depending upon non heated or heated bed. Experiment with whatever flexible sheet material you have locally for your top surface, the best is what is best for your specific applications - hence I have heated and non heated belt for different parts. Try a piece in your standard 3d printer and see what works.
I am now building a larger one with dual extruders - my application requires a carbon fibre reinforced Petg skin for strength and the foaming pla core material to reduce weight. It therefore copes with a non heated bed nicely. The carbon fibre minimises warping and the petg sticks well to the belt. The pla sticks well to the petg. The foaming pla gives a nice finish - shame it’s hidden inside.
The required printed part size is 1600 x 420 x 160 - so a 500mm belt width minimum. The 1/3 scale prototype worked well so fingers crossed on the full scale.