NAK3DDesigns/White-Knight #76: Ordering parts from Printed Solid

Vordir, 2022-01-15:
There are a few parts in the BOM I have a Problem with 

- Duet Wifi = There is only the new version "Duet2Wifi" available, but i red thats not a problem. is that right ?

- Mosquito Mount = is not available. but i saw in the CAD that this would be a "Groove Mount Bowden Adapter"  
   https://www.3djake.de/e3d/groove-mount-bowden-adapter ?

-Slice Thermistor = there isn´t stated which one i need. when its the "SLICE ENGINEERING HIGH TEMP THERMISTOR"  is it the availible "RTD PT 1000" for 50$ or would the 300°C version OK ? but this ins´t available either. is there an alternative ?
   https://www.printedsolid.com/products/slice-engineering-high-temp-thermistor-1

-Silce 50W heater = is it the "SLICE ENGINEERING 50W HEATER" in 12 V or 24 V version ? 
    https://www.printedsolid.com/collections/all/products/slice-engineering-50w-heater?variant=12533493989461

-LDO .9 Stepper = there is no such stepper on the website.  I only found this : https://www.ebay.de/itm/333367708304
   I dont know much about electronics. chose this project to learn something about it, but dont want to order the wrong parts

-Nema 17 5.1 geared stepper = on the stated website not found. found only 5:1 geared here https://www.ebay.de/itm/392172201987 
                                                    is that right ? is there an alternative 

Thanks in advance
Yours sincerely Daniel

NAK3DDesigns, 2022-01-15:
Yes, the Duet2Wifi is what you want.
I believe I used the 300°C
As for the heater if you go with the recommended 24V power supply then you need the 24V heater cartridge

Attached is an image from the printed solid shopping cart showing the 5:18 Geared stepper, the .9 steppers and the slice Mosquito groove mount.

Hope that helps

[image]

Vordir, 2022-01-17:
helped a lot, thanks :)

closed 2022-01-17 by Vordir