NAK3DDesigns/White-Knight #78: 3d printer

Ladyshakti, 2022-11-26:


Ladyshakti, 2022-11-26:
I can load the file into fusion 360 on my Android phone to see the instructions

Ladyshakti, 2022-11-26:
I can not load the file. I've downloaded the file in assembly for fusion 360 but it doesn't load. When I try other programs it says there is an character error on line 1.  I want to see the measurements for frame and what goes where. I also need it for roller including the extra space between belt and frame. If I can get this I can work out the new measurements for the print size I want. 1000mm height and 1500mm wide.  I am not looking for free advise and waste your time.  I intend on paying for your experience

tobitus, 2022-12-08:
Clicking "save as" on the file link doesn't work in github.

Here's how to view the file in fusion (this is on desktop):

go to this page:
https://github.com/NAK3DDesigns/White-Knight/tree/master/Assembly

Hover over the link "White Knight Full Assembly v126.f3z" and you will see a check box pop up to the left. 
Click on the checkbox so it turns blue.

Now look on the bottom right of your screen and you will see a black arrow pointing down.  Click that arrow.
-github will download the correct file.

no open this file from fusion and it will import all the pieces like it is supposed to.

MR-lwj, 2023-01-27:
Excuse me, Sir, I am a 3D printing enthusiast from China.I want you to know some questions about the control board of this printer. May I ask what type of motherboard you have?

NAK3DDesigns, 2023-01-30:
> Excuse me, Sir, I am a 3D printing enthusiast from China.I want you to know some questions about the control board of this printer. May I ask what type of motherboard you have?

It uses a Duet Wifi.  All components used are listed in the BOM.